SensorsINI/jaer #2: Can't see eDVS in USB interface list - librxtx problem?

Huitzilo, 2017-07-06:
Hi,
my eDVS camera which I've connected via USB doesn't show up in the eDVS hardware Interface chooser. When I hit refresh, the following is logged to the console:

`java.lang.NoClassDefFoundError: com/sun/jna/platform/win32/Win32Exception thrown while loading gnu.io.RXTXCommDriver`

System details: 
java version "1.8.0_131" (64 bit)
Ubuntu 16.04 LTS

Compiled jAER 1.5 and set up the udev rule as recommended in the docs. Setting chmod 0666 on /dev/ttyUSB0 didn't help. Tried it with eclipse as well - no luck. 

I installed librxtx and added the .so files to the classpath in the startup script - no change.

Any ideas?

Best regards,
Michael

federicohyo, 2017-07-06:
Hi,

have you tried to manually connect to the edvs as described here? 

https://inilabs.com/support/hardware/edvs/ see section "Accessing the device manually"

Can you actually see the device if you use a terminal connection?

Huitzilo, 2017-07-06:
Hi Frederico!

I had to sudo minicom although the permissions of /dev/ttyUSB0 are 0666. Does this mean there might be an issue with the udev rules?

Once minicom was running I tried entering `??` to see the help screen but to no avail. 

I had to set the baud rate to 12000000 via a config file, because minicom itself would only let me choose 45000000 at max.

tobidelbruck, 2017-07-07:
Hi Michael, did you by any chance try a windows machine? That seems to 
work more predictably with serial port for some users. Did you have a 
look at past posts? I know there was another thread about this recently.

Best, Tobi

Huitzilo, 2017-07-07:
Hi Tobi, 

it works under windows.

I had a look around the old forum. One thread suggested that using the latest git/svn snapshot under eclipse solved a somewhat related problem - will try this next and update the thread.

federicohyo, 2017-07-07:
Hi Michael,

Please also that now the edvs is now supported in libcaer and caer. These are c/c++ based softwares that can be used in linux and that targets embedded systems. 

Both these softwares are open source and available here: https://github.com/inilabs/

Note that you will need to download them from the github and compile them from source. They require  libserialport >= 1.1.1 to work with the eDVS.

Huitzilo, 2017-07-07:
Thanks Frederico, I'll give caer a try!

Huitzilo, 2017-07-12:
I managed to get caer working on ubuntu 16.04 and I can see the eDVS output in all its glory! ;) 
Still no idea why jAER fails on me though - any ideas how I could start to debug the serial port issue in Java?

giacomoi, 2017-07-12:
Hi Michael. We're going to try to get caer to work on Ubuntu 16.04 too in 
September. Was there any particular trick? Any libraries or undocumented 
packages that need to be installed?

Ciao,

   g.

Huitzilo, 2017-07-14:
Hi Giacomo!

I built a fairly minimal version of caerlib/caer, only with the visualiser and eDVS modules. The only hoop I had to jump through was that I had to install libserialport from source, as there is no .deb for 16.04 xenial with the required version. 
I should say the caer/libcaer cmake process seemed very well organised and made it a breeze resolving all dependencies. 👍

federicohyo, 2017-07-14:
Hi Michael,

thanks for the feedback. 
@giacomoi : note that libserialport is only needed for the eDVS module. 

best,
 Federico

gregburman, 2017-07-24:
Hi Everyone

I ran into this problem at the Telluride workshop, where I couldn't get the com port of the eDVS to show up in the Interface Chooser in jAER. This was the case on both Linux (Arch) as well as Windows. Eventually I decided to try rolling back to an earlier commit of jAER (on from the end of May) and this worked.

So some change in the jAER code base is the issue. git bisect should point you to the problematic changes that were made.

Huitzilo, 2017-07-24:
Hi Greg,
thanks for the hint, this probably means this is not just a problem on the ubuntu side.
Do you by any chance remember the exact commit that still worked for you?

llongi, 2017-08-03:
I have now reverted the nrjavaserial library to version 3.12.0. There seem to be issues on some systems with 3.13.0 (latest release). This should bring back at the least the status as it was in May 2017.

closed 2017-08-03 by llongi