SensorsINI/jaer #4: How to increase DAVIS240C update interval?

viktorbahr, 2017-08-14:
Hi :)

is there a way to decrease the inter-package interval?
So far I've played with the jAER options *FIFO buffer size* and *Number of Host buffer*, with no result (judging from the *USB statistics*).

Thanks,
Viktor

llongi, 2017-08-15:
Those won't necessarily help. What kind of interval are you seeing?
jAER does various levels of buffering. You can try to make the AERenderBufferSize smaller (USB menu -> Set rendering AE buffer size). Also try to change the USB maximum packet delay (Hardware Configuration -> Multiplexer Config tab -> USB.EarlyPacketDelay), but by default this is 8 (8x125µs slices, so 1ms), and that should really not be a problem. Getting less than 1ms is subject to jitter with USB, you can't reliably guarantee it.

viktorbahr, 2017-08-15:
Hi Luca,

thanks for the answer.
I just detected that we were processing things on rendering cycle instead of on aquisition cyle. Changing this reduced inter-packet interval to the desired sub-millisecond values.  
Sorry for the hassle..

---

For the sake of completeness, here's a few more details regarding our setup and the steps taken:

Current setup is as follows:  
 - *USB FIFO size* is 4096
 - 12 *Host USB read buffers*
 - *AEPacketRawBufferSize*  of 25000

Decreasing "USB.EarlyPacketDelay* from 8 to 2 did help with reducing average packet interval *variance*. Stablity seems not to be affected.

closed 2017-08-15 by viktorbahr