SensorsINI/jaer #6: package com.jogamp.opengl does not exist

asprasan, 2017-11-24:
For one of my projects I was trying to run the java program for extrapolating DAVIS frames. The code for that is available on the path src\ch\unizh\ini\jaer\projects\davis\frames. While compiling the code I got the error saying package com.jogamp.opengl does not exist. I checked the com folder in src and package jogamp doesn't exist. Has the package been moved to some other location? Or am I getting anything wrong? I'm newbie to java and don't quite understand well, how to run java codes.

tobidelbruck, 2017-11-24:
It sounds like you are not using an IDE that has the classpaths and 
JAVA_NATIVE_LIBRARY set up properly. Did you open either the netbeans or 
eclipse project? That should set up everything so you can just hit F6 to 
build and run from the working copy.

Please ask if this is still confusing. I know it is a bit murky the 
first time using a Java project.

Tobi

asprasan, 2017-11-24:
I have now installed Netbeans and I built, run the jaer project. Once I press F6, it opens me the jaerViewer GUI to view and log DVS/ DAVIS data. But, how do I run a particular code? When I run the particular code in Netbeans that I want (Shift+F6), it says 'ApsFrameExtrapolationISCAS does not have a main class'. Kindly help me with this.

tobidelbruck, 2017-11-24:
Please look at https://sourceforge.net/p/jaer/wiki/Home/, also 
https://inilabs.com/support/software/jaer/ . I believe there is also a 
Help menu that takes you to the user guide.

closed 2018-03-02 by tobidelbruck