SensorsINI/jaer #8: uncaughtException GLException after splash

harriganshane, 2018-01-22:
Hi.

I cloned the latest version of jAER from the github and it was working fine for most of last week, but all of a sudden I got the error message below every time the splash logo disappears and the program subsequently terminates.

================================================================================

Jan 22, 2018 9:06:31 AM net.sf.jaer.util.LoggingThreadGroup uncaughtException
WARNING: Thread[AWT-EventQueue-0,6,main]
Jan 22, 2018 9:06:32 AM net.sf.jaer.util.LoggingThreadGroup uncaughtException
WARNING: com.jogamp.opengl.GLException: AWT-EventQueue-0: createImpl ARB n/a but required, profile > GL2 requested (OpenGL >= 3.1). Requested: GLProfile[GL3bc/GL3bc.hw], current: 3.0 (Compat profile, compat[ES2], FBO, hardware) - 3.0 Mesa 17.2.2
	at jogamp.opengl.x11.glx.X11GLXContext.createImpl(X11GLXContext.java:440)
	at jogamp.opengl.GLContextImpl.makeCurrentWithinLock(GLContextImpl.java:765)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:648)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:586)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1279)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at jogamp.opengl.GLAutoDrawableBase.defaultDisplay(GLAutoDrawableBase.java:467)
	at com.jogamp.opengl.GLAutoDrawableDelegate.display(GLAutoDrawableDelegate.java:190)
	at net.sf.jaer.JAERViewer.<init>(JAERViewer.java:145)
	at net.sf.jaer.JAERViewer.<init>(JAERViewer.java:73)
	at net.sf.jaer.JAERViewer$2.run(JAERViewer.java:297)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue.access$500(Unknown Source)
	at java.desktop/java.awt.EventQueue$3.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$3.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

Loading version info from resource BUILDVERSION.txt
Version URL=jar:file:/home/sharrigan/ProjectsAndGits/jaer/jaer/dist/jAER.jar!/BUILDVERSION.txt

-- listing properties --
build.date=20180122 at 0906

llongi, 2018-01-22:
I'm getting the same thing on my Linux system since some time now, never had time to investigate. Seems related to some update concerning the open-source Mesa 3D driver stack.

tobidelbruck, 2018-01-22:
There's a bit of a cleanup to do as well, since some parts of jaer seem 
to rely on jogl-2.0, while the latest release (for a long time now) is 
jogl.2.3.1 (jsr-231).  So we need refactor quite some code to benefit 
from bug fixes. I cannot figure out how to replace GLU and GLUT, which 
seem to be missing in the latest jogl, and which are used extensively to 
draw text and things like circles in many jaer filters.
(This all came up in experiments to use ivy to source packages from maven)

llongi, 2018-01-22:
The GLU/GLUT classes should still be part of JOGL 2.3.1, which we are in fact using right now. You need the jogl-all and gluegen-rt JARs from the JOGL distribution.
The Maven dependency looks like this:
`<dependencies>
    <dependency>
      <groupId>org.jogamp.gluegen</groupId>
      <artifactId>gluegen-rt-main</artifactId>
      <version>2.3.1</version>
    </dependency>
    <dependency>
      <groupId>org.jogamp.jogl</groupId>
      <artifactId>jogl-all-main</artifactId>
      <version>2.3.1</version>
    </dependency>
  </dependencies>`
You should be able to figure out the proper Ivy lines from that.

closed 2018-03-02 by tobidelbruck