SensorsINI/jaer #9: jAER user guide "Parameter persistence filter properties"

sanfans, 2018-03-15:
In the [jAER user guide](https://inilabs.com/support/software/jaer/#h.x2hejinqj2jt), section "Parameter persistence" (the link point to that), the link present behind "[EventFilter2D.getPrefs()]" is broken (click on it). Searching for "EventFilter2D" in the github jaer repository, i found [this](https://github.com/SensorsINI/jaer/blob/master/src/net/sf/jaer/eventprocessing/EventFilter2D.java), but inside there is no function "getPrefs()".
Where is located or which name has now?

tobidelbruck, 2018-03-16:
Sorry about that broken link that pointed back to the original SVN repo on sourceforge. I updated it now to properly point to the base class EventFilter. The simplest way to see how to store and retrieve preference properties is to look at at any recent EventFilter, e.g. Speedometer or PatchMatchFlow. These both have many preference property values of various types, including (in PatchMatchFlow) enums and String values. 
Hope this helps. 
It is much easier to inspect the code in netbeans, since you can navigate variable uses by putting cursor inside a variable or method name highlights all the instances and using the alt-uparrow and alt-downarrow keys navigates between them. Maybe there is equivalent in eclipse.
Hope this helps

tobidelbruck, 2018-05-19:
closed, doc link fixed

closed 2018-05-19 by tobidelbruck