SensorsINI/jaer #10: Invalid Parameter

ardayigit, 2018-03-27:
On Ubuntu 17.10, when I try to connect to my eDVS (serial 126x), I have the following (12Mbps) error: **gnu.io.UnsupportedCommOperationException: Invalid Parameter**. 

I don't have this problem on Windows but what should I do to run jAER on Linux ? 

Here is the output on the jAER console: 
16:04:56 INFO: net.sf.jaer.graphics.AEViewer.<init>: AEViewer starting up...
16:04:56 INFO: net.sf.jaer.graphics.AEViewer.<init>: AEChip class name is ch.unizh.ini.jaer.chip.retina.DVS128
16:04:56 INFO: net.sf.jaer.graphics.AEViewer.<init>: using /tmp as the defaultLoggingFolderName
16:04:56 INFO: net.sf.jaer.graphics.AEViewer.<init>: AEViewer.lastLoggingFolder=/tmp
16:04:57 INFO: net.sf.jaer.hardwareinterface.serial.eDVS128.eDVS128_InterfaceFactory$InitializePortListThread.run: starting thread to initialize serial port list
16:04:57 INFO: net.sf.jaer.hardwareinterface.serial.eDVS128.eDVS128_InterfaceFactory.refreshSerialPortList: enumerating serial ports....
16:04:57 INFO: net.sf.jaer.graphics.AEViewer.buildInterfaceMenu: finding number of available interfaces
16:04:57 INFO: net.sf.jaer.hardwareinterface.serial.eDVS128.eDVS128_InterfaceFactory.refreshSerialPortList: done enumerating serial ports
16:04:57 INFO: net.sf.jaer.hardwareinterface.serial.eDVS128.eDVS128_InterfaceFactory$InitializePortListThread.run: serial port initialization thread done
16:04:57 WARNING: net.sf.jaer.graphics.AEViewer.getChipClassPrefs: Building list of default AEChip devices - this can takes some time. To reduce startup time, use AEChip/Customize to specify desired devices
16:04:57 INFO: net.sf.jaer.util.RemoteControl.<init>: Constructed uninitialized and empty RemoteControl on port=8995
16:04:57 INFO: net.sf.jaer.chip.Chip.<init>: Created RemoteControl on port=8995 for control of DVS128 sizeX=0 sizeY=0 eventClass=null
16:04:57 INFO: net.sf.jaer.graphics.ChipCanvas.<init>: GLCanvas=AWT-GLCanvas[Realized false,
	null-drawable,
	Factory   null,
	handle    0x0,
	Drawable size -1x-1 surface[200x200],
	AWT[pos 0/0, size 200x200,
	visible true, displayable false, showing false,
	null]]
16:04:57 INFO: net.sf.jaer.graphics.ChipCanvas.setDisplayMethod: no menu item for display method net.sf.jaer.graphics.ChipRendererDisplayMethod@6745758a cannot set it as the selected display method in the menu
16:04:57 INFO: eu.seebetter.ini.chips.davis.HotPixelFilter$HotPixelSet.loadPrefs: no hot pixels to load
16:04:57 WARNING: net.sf.jaer.graphics.AEFrameChipRenderer.<init>: cannot make a DavisVideoContrastController for this chip because it does not extend DavisChip
16:04:57 INFO: net.sf.jaer.chip.AEChip.onRegistration: registering DVS128 sizeX=128 sizeY=128 eventClass=PolarityEvent
16:04:57 INFO: net.sf.jaer.graphics.AEViewer$ViewLoop.run: sleeping until isVisible()==true
16:04:57 INFO: net.sf.jaer.eventio.AEServerSocket.run: bound AEServerSocket on port=8990
16:04:57 INFO: net.sf.jaer.util.RemoteControl.<init>: Constructed uninitialized and empty RemoteControl on port=8997
16:04:57 INFO: net.sf.jaer.graphics.AEViewer.<init>: created RemoteControl on port=8997 for remote control of some AEViewer functions
16:04:57 INFO: net.sf.jaer.util.WindowSaver.loadSettings: window y origin is < lowerInset, moving back to 0
16:04:57 INFO: net.sf.jaer.graphics.ChipCanvas.init: OpenGL implementation is: jogamp.opengl.gl4.GL4bcImpl
GL_VENDOR: Intel Open Source Technology Center
GL_RENDERER: Mesa DRI Intel(R) Haswell Mobile 
GL_VERSION: 3.0 Mesa 17.2.8
16:04:58 INFO: net.sf.jaer.util.WindowSaver.loadSettings: window y origin is < lowerInset, moving back to 0
16:05:02 INFO: net.sf.jaer.graphics.AEViewer.buildInterfaceMenu: finding number of available interfaces
16:05:06 WARNING: net.sf.jaer.hardwareinterface.serial.eDVS128.eDVS128_InterfaceFactory.doChooseSerial: Caught exception gnu.io.UnsupportedCommOperationException: Invalid Parameter; this can mean port is owned by another process

tobidelbruck, 2018-03-27:
It probably means that under linux, that baud rate is not supported 
either by driver or by the java library used in jAER.  Jorg Conradt is 
the real expert here but maybe Luca Longinotti at inivation can 
reproduce this problem since he has a linux setup to try it.   Have you 
tried putting a breakpoint in doChooseSerial to see the parameters being 
set?  Can you change the camera baud rate to something like 115200, 
which is supported for sure on all platforms? That would be very slow 
but would confirm the problem. Also did you check the serial library 
being used? They tend to be fragile and not very supported in java.

ardayigit, 2018-03-27:
How can I change the baudrate of my device? Can I do it on Linux? Is it enough to "Open serial port interface" with another Baud rate (1, 2, 4 or 8)? Or do I need to modify some configs? 

Thank you very much for your answer!

llongi, 2018-03-27:
I've never gotten the jAER Linux serial device support to work reliably. The library we use (nrjavaserial) has problems on Linux, at least with high baudrates. That's why I developed the libcaer eDVS integration, and have been using that ever since.

tobidelbruck, 2018-05-19:
closed due to inactivity;

closed 2018-05-19 by tobidelbruck