SensorsINI/jaer #17: DAVIS346 Blue not working properly(installed required version 16)

yucicheung, 2018-11-03:
We connect our DAVIS346 blue to computer, the jAER(1.7.1) told us 

> Device logic version incorrect. You have version 9912, but version 16 is required.

We updated our DAVIS346 Blue to version 16. The _AEChip_ is set as _Davis346blue_, the _HW configuration_  is loaded with file _DAVIS346blue.xml_.
However, the jAER does not show the image correctly. For instance, the reconstructed image does not present what's in front of the camera, instead it's full of noise(like the pic shown below).
[image]

How to solve this issue?
1. To downgrade this camera to 9912?
2. Or what else?

yucicheung, 2018-11-03:
Also, while using the latest libcaer, the error message

> Timestamps on host and device are drifting away

shows up.

llongi, 2018-11-03:
> Reply-to: SensorsINI/jaer
> <reply+0046590e82e10f016416d1f7cd3942f39e31d54f979ca62092cf0000000117f5
> 54c792a169ce16793264@reply.github.com>
> properly(installed required version 16) (#17)
>
> I'm not quite sure which one is the device name.
> To be more specific:
> The device name recognized by my computer is DAVIS346.
>
> As depicted in the picture, the AEChip is set as Davis346blue,so the
> bar of jAER writes Davis346blue - DAVIS346
>
> The bias settings are loaded from
> biasgenSettings/Davis346b/DAVIS346blue.xml.
> Which name should the renamed bias setting file keep consistency with?

The upgrade to version 16 looks fine.
The correct chip class is Davis346blue, so that's also fine.
And the file to load is indeed:
"biasgenSettings/Davis346b/DAVIS346blue.xml"
So it should be working properly.

Can you send us a screenshot of the 'Bias' configuration dialog before
and after you re-load the settings file?

Does it work properly in libcaer/cAER?

The 'timestamps are drifting away' message is normal after some time
for debug builds, debug builds have additional checks for us
developers. This one comparse the progression of time on the device
(crystal oscillator on device) versus the progression of time on your
host system (usually kept by your CPU, with syncing to some network
time source). Usually over time the CPU is more precise with regards to
what we as humans may call "real-time" (what the reference atomic
clocks around the world agree on), as it has more precise oscillators
and can compensate drift by syncing to network time sources (or GPS if
you want to be really precise).
This is something on our list of things to possibly improve by having
the host system advise the device on when to skip or stretch time to
sync up again, but it's very low on our priority list, as it hasn't
proven to be a problem for developing algorithms, where you usually
look at device timestamps always to establish some happens-before
relationship.
-- 
Luca Longinotti (llongi)

Software Engineer
iniVation AG - https://inivation.com/
Zurich, Switzerland
Office: +41 44 500 32 14

Support - https://inivation.com/support/

yucicheung, 2018-11-04:
Is there any chance that this problem is caused due to  the camera was connected to a USB2.0 port?

tobidelbruck, 2018-11-04:
> How do I confirm that the upgrade to version 16 is fine?

Open the device in flashy and I think there is both a user friendly info 
tab and a detailed info tab.

tobidelbruck, 2018-11-04:
I really doubt this is the case. What is the view from flashy now of the 
device firmware and logic? Is it possible you burned the incorrect FPGA 
logic file for a different camera?

yucicheung, 2018-11-04:
I took a picture of the flashy interface after upgrade to logic version 6.
[image]

Here's how the device is upgraded(following the instructions in `automatic upgrade`  part from [Reflashing Guide](https://inivation.com/support/software/reflashing/#automatic-upgrade)):
1. Open flashy1.2.0.
2. We saw an button titled `update logic` and clicked the button
3. Falshy informed us it's successfully updated.
4. Unplug and replug the device and we saw the above picture.

Here's a question to @tobidelbruck .
>  git pull the biasgenSettings folder and load the renamed biases from
Davis346Blue.xml.
> When the device name changed, all the preference keys also changed. You
just need to load these settings from this file with the new AEChip name.

Does this anwser mean that  I need to rename any file to make the camera work properly?

yucicheung, 2018-11-04:
If We've used the `automatic upgrade`  method to update the logic version, is it possible that it will burn the incorrect FPGA logic file for a different camera?

Is this a circumstance that we should try manually update/reflash the device.

tobidelbruck, 2018-11-04:
No, the keys in the jaer preferences file correspond to the AEChip name 
selected in AEViewer AEChip menu. You can just load this file to import 
these key/value pairs to the java preferences registry.

git pull the biasgenSettings folder and load the renamed biases from
>
>     Davis346Blue.xml.
>     When the device name changed, all the preference keys also
>     changed. You
>     just need to load these settings from this file with the new
>     AEChip name.
>
> Does this anwser mean that I need to rename any file to make the 
> camera work properly?
>

yucicheung, 2018-11-05:
I know it's not recommended, but considering the compatibility with widely-used jAER(1.6.x), I think  downgrading the logic version back to 9912 is a safe method to make the camera work.

So I want to know is there logic 9912 in the commit history of https://github.com/inivation/devices-bin/tree/master/logic ? If there is, would you mind helping me locate the file?

I appreciate a lot!

yucicheung, 2018-11-05:
When I tried to flash my device back to logic 9912 using Flashy1.2.0, I selected logic 9912 file and clicked `upload logic bitstream` button.
I got  an _Exception_ said:
> Unable to send vendor OUT request 0xbe: LIBUSB_ERROR_PIPE
and  an USB error massage said:
>Error message：’System started.‘(code 0 at time - 1509359616)
>Error message：’VR_CPLD_UPLOAD INIT: unsupported CPLD ID'(code 74 at time 1133314048).

The screenshot is shown as below.
[image]

1. Is is a prove that our reflashing to logic 16 was not successful?
2. How do I fix the camera?

llongi, 2018-11-05:
Please use the Flash Logic button. In theory the upload logic button should be disabled so that may be a bug in Flashy.

[exception]<https://user-images.githubusercontent.com/11587022/47993211-e82d6d00-e129-11e8-81a2-e02f937181a2.PNG>

yucicheung, 2018-11-06:
I’ve downgraded the logic to 9912.
The camera is now able to work properly.

Thank you @tobidelbruck  and @llongi  for your help!

closed 2018-11-06 by yucicheung

llongi, 2018-11-21:
Just a quick update: I've managed to find the issue with the logic update and the blue DAVIS.
There will be new firmware (v5) and logic (v18) released in the course of the next week to fix this.
jAER and cAER will also have new releases that require those versions at that time.