SensorsINI/jaer #20: Not compatible with java se 11

federicohyo, 2019-02-22:
Required version of Java SE is 8u.

It will not compile with newer versions of java... see log below.

ant -f C:\\Users\\corrad61\\Documents\\Software\\jaer -Dnb.internal.action.name=rebuild clean jar
download-deps:
:: Apache Ivy 2.4.0 - 20141213170938 :: http://ant.apache.org/ivy/ ::
:: loading settings :: file = C:\Users\corrad61\Documents\Software\jaer\ivysettings.xml
:: resolving dependencies :: org.jaerproject#jaer;working@ICT0117049brd
	confs: [default]
	found org.jogamp.gluegen#gluegen-rt;2.3.2 in public
	found org.jogamp.gluegen#gluegen-rt-main;2.3.2 in public
	found org.jogamp.jogl#jogl-all-main;2.3.2 in public
	found org.jogamp.jogl#jogl-all;2.3.2 in public
	found org.swinglabs#swing-layout;1.0.3 in public
	found org.jdesktop#beansbinding;1.2.1 in public
	found com.github.swri-robotics#bag-reader-java;1.10.0 in public
	found com.google.guava#guava;18.0 in public
	found org.slf4j#slf4j-api;1.7.16 in public
	found commons-codec#commons-codec;1.10 in public
	found com.fasterxml.jackson.core#jackson-databind;2.6.5 in public
	found com.fasterxml.jackson.core#jackson-annotations;2.6.0 in public
	found com.fasterxml.jackson.core#jackson-core;2.6.5 in public
	found org.apache.commons#commons-compress;1.10 in public
	found com.github.swri-robotics#lz4;1.5.0 in public
	found com.google.code.findbugs#jsr305;1.3.9 in public
	found ch.qos.logback#logback-classic;1.1.5 in public
	found ch.qos.logback#logback-core;1.1.5 in public
	found org.apache.commons#commons-lang3;3.7 in public
	found org.apache.commons#commons-math3;3.6.1 in public
	found org.webjars.npm#semver;5.4.1 in public
	found commons-net#commons-net;3.6 in public
	found org.apache.directory.studio#org.apache.commons.io;2.4 in public
	found commons-io#commons-io;2.4 in public
	found findbugs#annotations;1.0.0 in public
	found com.google.code.gson#gson;2.8.2 in public
	found javax.mail#javax.mail-api;1.6.0 in public
	found org.jblas#jblas;1.2.4 in public
	found org.scala-saddle#jhdf5;2.9 in public
	found org.scala-lang#scala-library;2.9.2 in public
	found org.bytedeco.javacpp-presets#hdf5;1.10.1-1.4 in public
	found org.bytedeco#javacpp;1.4 in public
	found net.java.linoleum#jlfgr;1_0 in public
	found com.neuronrobotics#nrjavaserial;3.13.0 in public
	found net.java.dev.jna#jna;4.2.2 in public
	found net.java.dev.jna#jna-platform;4.2.2 in public
	found com.kitfox.svg#svg-salamander;1.0 in public
	found org.uncommons.maths#uncommons-maths;1.2.2a in public
	found jfree#jcommon;1.0.12 in public
	found jfree#jfreechart;1.0.8a in public
	found org.tallison#jmatio;1.2 in public
	found org.usb4java#usb4java;1.2.0 in public
	found org.usb4java#libusb4java;1.2.0 in public
	found org.bytedeco.javacpp-presets#opencv;3.2.0-1.3 in public
	found org.openpnp#opencv;3.2.0-1 in public
	found org.bytedeco#javacv;1.0 in public
	found org.bytedeco.javacpp-presets#ffmpeg;2.7.1-1.0 in public
	found org.bytedeco.javacpp-presets#flycapture;2.7.3.19-1.0 in public
	found org.bytedeco.javacpp-presets#libdc1394;2.2.3-1.0 in public
	found org.bytedeco.javacpp-presets#libfreenect;0.5.2-1.0 in public
	found org.bytedeco.javacpp-presets#videoinput;0.200-1.0 in public
	found org.bytedeco.javacpp-presets#artoolkitplus;2.3.1-1.0 in public
	found org.bytedeco.javacpp-presets#flandmark;1.07-1.0 in public
	found com.google.android#android;2.2.1 in public
	found org.jogamp.jocl#jocl-main;2.0.2 in public
	found org.jogamp.jocl#jocl;2.0.2 in public
	found junit#junit;4.11 in public
	found org.hamcrest#hamcrest-core;1.3 in public
	found org.tensorflow#tensorflow;1.12.0 in public
	found org.tensorflow#libtensorflow;1.12.0 in public
[... 367 lines cut ...]
  symbol:   variable DatatypeConverter
  location: class EasyXMLReader
C:\Users\corrad61\Documents\Software\jaer\src\ch\unizh\ini\jaer\projects\multitracking\ZebrafishMedianTracker.java:122: error: cannot find symbol
            Color color=Color.hsb(hue, 0.5f, 1f);
  symbol:   class Color
  location: class ZebrafishMedianTracker
C:\Users\corrad61\Documents\Software\jaer\src\ch\unizh\ini\jaer\projects\multitracking\ZebrafishMedianTracker.java:122: error: cannot find symbol
            Color color=Color.hsb(hue, 0.5f, 1f);
  symbol:   variable Color
  location: class ZebrafishMedianTracker
Note: Some input files additionally use or override a deprecated API.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
6 errors
100 warnings
BUILD FAILED (total time: 48 seconds)

######################################
According to the release-notes, Java 11 removed the Java EE modules:

java.xml.bind (JAXB) - REMOVED
Java 8 - OK
Java 9 - DEPRECATED
Java 10 - DEPRECATED
Java 11 - REMOVED

tobidelbruck, 2019-02-22:
Thanks Federico. I will investigate. The current situation with Java 11 
and Netbeans is a bit of a mess since Java and the Oracle libraries got 
a major facelift, a big change in licensing, and Netbeans got moved to 
Apache where the usability is currently not so great, at least for now.

The recommended path right now is to just download jdk 1.81+netbeans 8.2 
bundle and everything will just work. We just did this yesterday.  This 
bundle is a bit hard to find on oracle but here is the link (at least 
for now)

https://www.oracle.com/technetwork/java/javase/downloads/jdk-netbeans-jsp-3413139-esa.html

Tobi

tobidelbruck, 2024-01-12:
This issue is resolved with jaer-2.0.2

closed 2024-01-12 by tobidelbruck