SensorsINI/jaer #23: java.lang.IllegalStateException: no splash screen available

Rajawat23, 2019-05-13:
Before describing the problem in detail, following relevant information.
Release:- jaer-1.7.2
Java version:- 12.0.1
Camera:- DVS128
System overview:- System Version: macOS 10.14.4 (18E226)
                                 Kernel Version: Darwin 18.5.0

**Issue**
After logging and saving the `.aedat` file when I try to open them on the application, the applications exits with the above-mentioned error.

tobidelbruck, 2019-05-13:
Sorry about this problem. We cannot debug on MacOS, so difficult to 
help.  Does the splash screen problem occur even without opening a file? 
There should not be any splash screen displayed when opening a file from 
within the AEViewer if it is already running.

Another thing you can try is to erase your jAER Java preferences. Search 
web or see e.g. 
https://alvinalexander.com/java/java-preferences-mac-os-x-where

Rajawat23, 2019-05-13:
No, it records perfectly, even replays after recording but once I save it, and want to open it. The video plays successfully in preview but it throws the error as soon as clicked on `Open` button

tobidelbruck, 2019-05-13:
You could try to avoid this preview by opening directly from File/Recent 
Files pulldown?  I don't know why the IllegalStateException is thrown. 
If you can capture the logging output (by running jAER from command 
line, for example) maybe I could track down the offending call and tell 
you how to comment something out. But I can't debug it here.

Rajawat23, 2019-05-13:
Hey, @tobidelbruck 
opening from recent files it works. 
Although I have been running jAER from the command line. 
For your reference, pasting the console output and attaching the error log generated by it.

**Console output**
```
./jAERViewer_linux.sh
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.jogamp.common.os.NativeLibrary$3 (file:/Users/aks/Documents/jaer/jaer-dist/lib/gluegen-rt-2.3.2.jar) to method java.lang.ClassLoader.findLibrary(java.lang.String)
WARNING: Please consider reporting this to the maintainers of com.jogamp.common.os.NativeLibrary$3
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
2019-05-13 15:17:06.047 java[17560:954042] WARNING: NSWindow drag regions should only be invalidated on the Main Thread! This will throw an exception in the future. Called from (
	0   AppKit                              0x00007fff33868e63 -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 378
	1   AppKit                              0x00007fff33866253 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1479
	2   AppKit                              0x00007fff33865c86 -[NSWindow initWithContentRect:styleMask:backing:defer:] + 45
	3   libnativewindow_macosx.jnilib       0x000000011a3c83fe Java_jogamp_nativewindow_macosx_OSXUtil_CreateNSWindow0 + 398
	4   ???                                 0x000000011abc9990 0x0 + 4743534992
)
JAERViewer.SplashHandler.publish(): splash screen is null or no longer visible, closing logging to it
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x000000011a2bdb80, pid=17560, tid=775
#
# JRE version: Java(TM) SE Runtime Environment (12.0.1+12) (build 12.0.1+12)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (12.0.1+12, mixed mode, sharing, tiered, compressed oops, g1 gc, bsd-amd64)
# Problematic frame:
# C  [libosxapp.dylib+0x2b80]  -[NSApplicationAWT sendEvent:]+0x176
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /Users/aks/Documents/jaer/jaer-dist/hs_err_pid17560.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
./jAERViewer_linux.sh: line 15: 17560 Abort trap: 6           java -classpath "dist/jAER.jar:jars/*:lib/*" -splash:images/SplashScreen.gif -Djava.util.logging.config.file=conf/Logging.properties -Dsun.java2d.noddraw=true -Dsun.java2d.opengl=false net.sf.jaer.JAERViewer
```
**error log**
[hs_err_pid17560.log](https://github.com/SensorsINI/jaer/files/3173043/hs_err_pid17560.log)

tobidelbruck, 2019-05-13:
Thanks, that's helpful. Maybe related to this thread: 
http://forum.jogamp.org/How-to-enable-Gluegen-Java3D-on-Java-9-td4038338.html

It looks to be a bug buried in JOGL for MacOS...  maybe you can use a 
different JRE.

Rajawat23, 2019-05-14:
Thanks 
Closing the issue 👍

closed 2019-05-14 by Rajawat23