SensorsINI/jaer #25: Can't use dvs-slice-avi-writer

Drallert, 2019-06-12:
Hey there, I'm trying to use said binary to convert videos I have recorded in .aedat format to .avi, but I haven't been able yet to do it.
I've tried the windows executable specifying all options and everything I end up getting is always the same, the help displayed by the command when you run it incorrectly (I guess, I have no idea really of what's going on), here are some of my attemps to make it work:
```
Microsoft Windows [Versión 10.0.17134.829]
(c) 2018 Microsoft Corporation. Todos los derechos reservados.

C:\Users\enriq\Desktop\Mis cosas\Universidad\TFG\jAER>dvs-slice-avi-writer.exe D:\video2-2019-04-04T15-54-47+0200-000022IL-0.aedat
Writing to output file D:\video2-2019-04-04T15-54-47+0200-000022IL-0.avi
constructing AEChip eu.seebetter.ini.chips.davis.Davis346B
java.lang.IllegalArgumentException: OptionSet: unknown key: dimx
        at ml.options.OptionSet.isSet(OptionSet.java:94)
        at net.sf.jaer.util.avioutput.DvsSliceAviWriter.main(DvsSliceAviWriter.java:385)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
        at com.exe4j.runtime.LauncherEngine.launch(LauncherEngine.java:81)
        at com.exe4j.runtime.WinLauncher.main(WinLauncher.java:94)

C:\Users\enriq\Desktop\Mis cosas\Universidad\TFG\jAER>dvs-slice-avi-writer.exe
java DvsSliceAviWriter [-aechip=aechipclassname (fully qualified class name, e.g. eu.seebetter.ini.chips.davis.DAVIS240C)] [-dvsSubsampler.getWidth()=36] [-dvsSubsampler.getHeight()=36] [-quality=.9] [-format=PNG|JPG|RLE|RAW] [-framerate=30] [-grayscale=200] [-writedvssliceonapsframe=false] [-writetimecodefile=true] [-numevents=2000] [-rectify=false] [-normalize=true] [-showoutput=true]  [-maxframes=0] inputFile.aedat [outputfile.avi]
Note arguments values are assigned with =, not space
If outputfile is not provided its name is generated from the input file with appended .avi

C:\Users\enriq\Desktop\Mis cosas\Universidad\TFG\jAER>dvs-slice-avi-writer.exe -dvsSubSampler.getWidth()=640 D:\video2-2019-04-04T15-54-47+0200-000022IL-0.aedat
java DvsSliceAviWriter [-aechip=aechipclassname (fully qualified class name, e.g. eu.seebetter.ini.chips.davis.DAVIS240C)] [-dvsSubsampler.getWidth()=36] [-dvsSubsampler.getHeight()=36] [-quality=.9] [-format=PNG|JPG|RLE|RAW] [-framerate=30] [-grayscale=200] [-writedvssliceonapsframe=false] [-writetimecodefile=true] [-numevents=2000] [-rectify=false] [-normalize=true] [-showoutput=true]  [-maxframes=0] inputFile.aedat [outputfile.avi]
Note arguments values are assigned with =, not space
If outputfile is not provided its name is generated from the input file with appended .avi

C:\Users\enriq\Desktop\Mis cosas\Universidad\TFG\jAER>dvs-slice-avi-writer.exe [-dvsSubSampler.getWidth()=640] D:\video2-2019-04-04T15-54-47+0200-000022IL-0.aedat
Warning: Input filename does not end with aedat: [-dvsSubSampler.getWidth()=640]
constructing AEChip eu.seebetter.ini.chips.davis.Davis346B
java.lang.IllegalArgumentException: OptionSet: unknown key: dimx
        at ml.options.OptionSet.isSet(OptionSet.java:94)
        at net.sf.jaer.util.avioutput.DvsSliceAviWriter.main(DvsSliceAviWriter.java:385)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
        at com.exe4j.runtime.LauncherEngine.launch(LauncherEngine.java:81)
        at com.exe4j.runtime.WinLauncher.main(WinLauncher.java:94)

C:\Users\enriq\Desktop\Mis cosas\Universidad\TFG\jAER>dvs-slice-avi-writer.exe -dvsSubSampler.getWidth()=640 D:\video2-2019-04-04T15-54-47+0200-000022IL-0.aedat
java DvsSliceAviWriter [-aechip=aechipclassname (fully qualified class name, e.g. eu.seebetter.ini.chips.davis.DAVIS240C)] [-dvsSubsampler.getWidth()=36] [-dvsSubsampler.getHeight()=36] [-quality=.9] [-format=PNG|JPG|RLE|RAW] [-framerate=30] [-grayscale=200] [-writedvssliceonapsframe=false] [-writetimecodefile=true] [-numevents=2000] [-rectify=false] [-normalize=true] [-showoutput=true]  [-maxframes=0] inputFile.aedat [outputfile.avi]
Note arguments values are assigned with =, not space
If outputfile is not provided its name is generated from the input file with appended .avi

C:\Users\enriq\Desktop\Mis cosas\Universidad\TFG\jAER>dvs-slice-avi-writer.exe dvsSubSampler.getWidth()=640 D:\video2-2019-04-04T15-54-47+0200-000022IL-0.aedat
Warning: Input filename does not end with aedat: dvsSubSampler.getWidth()=640
constructing AEChip eu.seebetter.ini.chips.davis.Davis346B
java.lang.IllegalArgumentException: OptionSet: unknown key: dimx
        at ml.options.OptionSet.isSet(OptionSet.java:94)
        at net.sf.jaer.util.avioutput.DvsSliceAviWriter.main(DvsSliceAviWriter.java:385)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
        at com.exe4j.runtime.LauncherEngine.launch(LauncherEngine.java:81)
        at com.exe4j.runtime.WinLauncher.main(WinLauncher.java:94)

C:\Users\enriq\Desktop\Mis cosas\Universidad\TFG\jAER>dvs-slice-avi-writer.exe -dvsSubSampler.getWidth()=640 D:\video2-2019-04-04T15-54-47+0200-000022IL-0.aedat
java DvsSliceAviWriter [-aechip=aechipclassname (fully qualified class name, e.g. eu.seebetter.ini.chips.davis.DAVIS240C)] [-dvsSubsampler.getWidth()=36] [-dvsSubsampler.getHeight()=36] [-quality=.9] [-format=PNG|JPG|RLE|RAW] [-framerate=30] [-grayscale=200] [-writedvssliceonapsframe=false] [-writetimecodefile=true] [-numevents=2000] [-rectify=false] [-normalize=true] [-showoutput=true]  [-maxframes=0] inputFile.aedat [outputfile.avi]
Note arguments values are assigned with =, not space
If outputfile is not provided its name is generated from the input file with appended .avi

C:\Users\enriq\Desktop\Mis cosas\Universidad\TFG\jAER>dvs-slice-avi-writer.exe -dvsSubSampler.getWidth()=640 --dvsSubSampler.getHeight()=640 D:\video2-2019-04-04T15-54-47+0200-000022IL-0.aedat
java DvsSliceAviWriter [-aechip=aechipclassname (fully qualified class name, e.g. eu.seebetter.ini.chips.davis.DAVIS240C)] [-dvsSubsampler.getWidth()=36] [-dvsSubsampler.getHeight()=36] [-quality=.9] [-format=PNG|JPG|RLE|RAW] [-framerate=30] [-grayscale=200] [-writedvssliceonapsframe=false] [-writetimecodefile=true] [-numevents=2000] [-rectify=false] [-normalize=true] [-showoutput=true]  [-maxframes=0] inputFile.aedat [outputfile.avi]
Note arguments values are assigned with =, not space
If outputfile is not provided its name is generated from the input file with appended .avi

[... 9 lines cut ...]
        at java.lang.reflect.Method.invoke(Unknown Source)
        at com.exe4j.runtime.LauncherEngine.launch(LauncherEngine.java:81)
        at com.exe4j.runtime.WinLauncher.main(WinLauncher.java:94)

C:\Users\enriq\Desktop\Mis cosas\Universidad\TFG\jAER>dvs-slice-avi-writer.exe -dvsSubSampler.getWidth()=640 -dvsSubSampler.getHeight()=480 D:\video2-2019-04-04T15-54-47+0200-000022IL-0.aedat
java DvsSliceAviWriter [-aechip=aechipclassname (fully qualified class name, e.g. eu.seebetter.ini.chips.davis.DAVIS240C)] [-dvsSubsampler.getWidth()=36] [-dvsSubsampler.getHeight()=36] [-quality=.9] [-format=PNG|JPG|RLE|RAW] [-framerate=30] [-grayscale=200] [-writedvssliceonapsframe=false] [-writetimecodefile=true] [-numevents=2000] [-rectify=false] [-normalize=true] [-showoutput=true]  [-maxframes=0] inputFile.aedat [outputfile.avi]
Note arguments values are assigned with =, not space
If outputfile is not provided its name is generated from the input file with appended .avi

C:\Users\enriq\Desktop\Mis cosas\Universidad\TFG\jAER>dvs-slice-avi-writer.exe
java DvsSliceAviWriter [-aechip=aechipclassname (fully qualified class name, e.g. eu.seebetter.ini.chips.davis.DAVIS240C)] [-dvsSubsampler.getWidth()=36] [-dvsSubsampler.getHeight()=36] [-quality=.9] [-format=PNG|JPG|RLE|RAW] [-framerate=30] [-grayscale=200] [-writedvssliceonapsframe=false] [-writetimecodefile=true] [-numevents=2000] [-rectify=false] [-normalize=true] [-showoutput=true]  [-maxframes=0] inputFile.aedat [outputfile.avi]
[... cut ...]
Error: no se ha encontrado o cargado la clase principal net.sf.jaer.util.avioutput.DvsSliceAviWriter
Causado por: java.lang.ClassNotFoundException: net.sf.jaer.util.avioutput.DvsSliceAviWriter
```
I have not changed anything, I just tried to run the programs as they came. I'm not sure if i have to do some kind of previous installation to get this working or whatever, but I'd really like to know how to get it going, as it's a very important part of the project I'm working on. 

Thank you very much in advance

tobidelbruck, 2019-06-13:
Hi Enrique, thanks for reporting this. The DvsSliceAviWriter class aged 
out from other changes due to internal changes. I just pushed a bunch of 
fixes to get it to work again.

Try this command line or some variation set as the startup option in 
netbeans after pull and rebuild:

-aechip=davis346blue -timeslicemethod=TimeIntervalUs 
-framedurationus=2000 -width=160 -height=120 -rectify=false 
-writeapsframes=false -writedvsframes=true -showoutput=true 
-maxframes=1000 -quality=1 -grayscale=100 -normalize=true -format=PNG 
-framerate=30   Davis346Blue-sample.aedat

You can try it from netbeans using the dvs-slice-avi-writer config 
option ( you might need to customize it further). The file needs to be 
in the jAER startup folder (the top level folder with the .exe launchers)

Drallert, 2019-06-13:
Hello Tobi,
Thank you very very much for your hastiness, I tried a couple of sets of options after rebuilding with netbeans and I finally got it working. I tried both video conversions separately (aps frames and dvs frames) and they worked out perfectly.
Here are the options I used:
```
dvs-slice-avi-writer.exe -aechip=eu.seebetter.ini.chips.davis.Davis346B -timeslicemethod=TimeIntervalUs 
-framedurationus=2000 -width=346 -height=260 -rectify=false -writeapsframes=false 
-writedvsframes=true -showoutput=true -maxframes=10000 -quality=1 -grayscale=100 normalize=true 
-format=PNG -framerate=30 "F:\video2-2019-04-04T15-54-47+0200-000022IL-0.aedat"

```
The one for the aps frames is the exact same one but with -writeapsframes=true and -writedvsframes=false

I'm closing the issue now and, again, thank you very much for your help.

closed 2019-06-13 by Drallert