SensorsINI/jaer #27: How to get a continuous set of .aedat file with a fixed logging time?

qwn19970728, 2019-11-19:
For example:I want to log and deal with event data every 10ms, how can i do it?Thank you!

tobidelbruck, 2019-11-19:
If you mean live data from sensor, then set the rendering rate to 100Hz 
by View/Set Rendering Rate... to 100Hz and then the main 
animation/processing loop will run this fast, if it can keep up. Each 
10ms you will get all the data that has been accumulated from USB driver 
since last call.  You might want to also disable View/Graphics 
options/Active Rendering enabled to allow coalescing of the rendering.

qwn19970728, 2019-11-19:
> If you mean live data from sensor, then set the rendering rate to 100Hz by View/Set Rendering Rate... to 100Hz and then the main animation/processing loop will run this fast, if it can keep up. Each [...]

So can I get the events.txt or events.jpg every 10ms?

tobidelbruck, 2019-11-19:
Wouldn't make much sense to do this.  Events.txt comes from a filter 
that writes this file out continuously.  Events.jpg output does not 
exist.  If you don't want to code java and prefer python, then use 
pyaer. jAER was intended for experiments in algorithmic real time event 
processing. It can stream data it collects over UDP if you want to use 
it as a front end. Check out the user guide please.

qwn19970728, 2019-11-19:
> Wouldn't make much sense to do this.  Events.txt comes from a filter that writes this file out continuously.  Events.jpg output does not exist.  If you don't want to code java and prefer python, [...]

OK,thank you!

qwn19970728, 2019-11-20:
> Wouldn't make much sense to do this.  Events.txt comes from a filter that writes this file out continuously.  Events.jpg output does not exist.  If you don't want to code java and prefer python, [...]

After your advice,I try to get the UDP data from jAER via python,but I get data like this:
'\x00\xb2\x81"!GX\x00\xf8X/\xc8\x1dE\x18\x00\xf8X5V\x08J\x98\x00\xf8X8H\x1fA8\x00\xf8X:\xe4\x1b\x08X\x00\xf8X;@\xac\xc0\x05\xdf\xf8X>4\xac\x80\x05\xe5\xf8X>4\xac@\x05\xe4\xf8X>4\xac\x00\x05\xe0\xf8X>4\xab\xc0\x05\xe0\xf8X>4\xab\x80\x05\xe3\xf8X>4\xab@\x05\xe6\xf8X>4\xab\x00\x05\x'

I'm not familiar with UDP, can you tell me how to decode this data to (x,y,t,p) event? Thank you!

tobidelbruck, 2019-11-20:
It is 'documented' in the control panel to set up the UDP output. You 
can also look at scripts that recieve and decode this binary stream.

I added notes to the jAER user guide as follows (in the UDP section)

**

*See scripts in the (non-maintained) sourceforge code repo 
https://sourceforge.net/p/jaer/code/HEAD/tree/scripts/matlab/. In 
particular, readudpstream.m 
<https://sourceforge.net/p/jaer/code/HEAD/tree/scripts/matlab/readaeudpstream.m>continuously 
reads a UDP data stream of a particular configuration (see code).*

*

To decode the raw data can be complicated, especially in case of DAVIS 
that outputs IMU data and APS frames. If the stream consists of only DVS 
events, then see code in 
https://sourceforge.net/p/jaer/code/HEAD/tree/scripts/matlab/retina/for 
DVS128 cameras, and in 
https://sourceforge.net/p/jaer/code/HEAD/tree/scripts/matlab/davis/for 
DAVIS cameras.

*

tobidelbruck, 2024-01-12:
Closed after no activity

closed 2024-01-12 by tobidelbruck