SensorsINI/jaer #28: No response and stopped in the starting picture

Rainyfish, 2019-11-28:
Hi,
    I tried many times to fix the environment of JVM, but jaer still had no response  and stopped in its starting picture. And the last comments appeared in picture is "Starting with no arguments in working directory = ." Windows 64 bit
   Looking forward to your reply.

tobidelbruck, 2019-11-28:
Did you watch YouTube setup video? Are you using jdk 1.8 ? Jdk 10 is not so tested. Please provide more details. 
Sorry this is from phone.

Rainyfish, 2019-11-29:
Thank you for your reply. I watched the setup video and the user guide.  I used JDK 1.8. Does the jaer need some special configs. The jaer works well in another computer. But it cannot run in other two computer even I used the same configs.

tobidelbruck, 2024-01-12:
Cause unknown, suggest to try bundled install4j installer which comes with Java JDK bundled.

closed 2024-01-12 by tobidelbruck