SensorsINI/jaer #29: aedat4 support

Tobias-Fischer, 2020-04-10:
Hi there,

I am wondering whether you plan to support the new aedat4 file format?

Many thanks, Tobias

tobidelbruck, 2020-04-10:
AEDAT-3 or AEDAT-4 would be great, but will require a major refactoring 
of many core classes of jAER. The problem is that when jAER started with 
DVS we didn't anticipate that the stream might included stuff besides 
AER events from various sensors, where the entirety of the data was by 
timestamp/address (each 32 bits). Thus internally the raw data packet 
format is such int arrays.

After the hardware interface or the AEFileInputStream, almost 
immediately the main animation loop cooks this RawPacket to a list of 
objects extending "BasicEvent".

If I had to do it over again I would use rosbags or some other generic 
container type (but not HDF5; too fragile to corruption).

For now the best short term way forward would be a python script that 
extracts DVS events (and maybe other things like IMU data and APS 
frames) from the AEDAT-4 file and writes them to the simple AEDAT-2 file 
format.  As far as I know, the utilities to work with AEDAT-3 and 
AEDAT-4 files are pretty much non-existent.

But I'm open to ideas here.

Tobias-Fischer, 2020-04-11:
I see, thanks for letting me know! Is there a Python script to write AEDAT-2 files from an event stream? Reading AEDAT-4 in python is trivial.

tobidelbruck, 2020-04-12:
Maybe in https://github.com/SensorsINI/processAEDAT/tree/master/jAER_utils ?

There are a LOT of scripts in this repo, but it is not very well 
organized or maintained now.  The matlab ones in misc date back to the 
earliest days of DVS when we developed an entire 
viewer/recording/playback framework in matlab for the DVS128, e.g. 
demo'ed at NIPS 2005 or 2006. The python stuff is more recent. People 
are certainly welcome to fork this and suggest pushes to help make it 
all better.

tobidelbruck, 2024-01-12:
See https://github.com/SensorsINI/aedat4to2 to convert files to jAER AEDAT-2.0 format

closed 2024-01-12 by tobidelbruck