SensorsINI/jaer #30: Step backwards not working

julio-guillen, 2020-06-10:
Hi,

I've noticed that every time I want to move in steps, using STEP FORWARD works fine, but STEP BACKWARDS also moves forward.

I'm using jAER 1.7.4. 

Thanks in advance :)

julio-guillen, 2020-06-10:
Hi,

I've also noticed that STEP BACKWARDS not only steps forward but it also **ignores my filter**.

Here we can see my filter is working. Its a simple filter that will mark all events outside a box as setFilteredOut(true)

https://pasteboard.co/JcqfrDV.png

And here, we can see a STEP BACKWARDS, where my filter wont work and no event will be filtered out:

https://pasteboard.co/JcqgnPs.png

Thanks in advance

tobidelbruck, 2020-06-10:
Yes, going backwards in time, all filters are automatically disabled 
because many of them break with nonmontonic timesstamps.  You can modify 
this behavior if you like, but it was causing a lot of problems to have 
it enabled.

julio-guillen, 2020-06-10:
Thanks :)

So, STEP BACKWARDS is disabled?

Best regards

tobidelbruck, 2020-06-11:
It's turns out not so easy to add this "step backwards" action and now I 
recall why it was not really implemented. I recommend you use "mark in" 
and "mark out" to loop forwards over your test data.

You can also use the rewind N packets ctrl-shift-left to go back some 
number of packets.

The problem is that step backwards is not like stepping back to the 
previous frame in conventional video. The desired action is that the 
time point is moved back by one frame (either in time or event count) 
and then a frame is played through the filter. But after this, you want 
the time to again be back at the starting time, before the packet is 
played, so that the next step back plays even more backwards. This will 
mess up any filters that expect monotonic time ordering.

julio-guillen, 2020-06-11:
Nevermind, Toby.

Thank you VERY much :)

julio-guillen, 2020-07-09:
Issue fixed in https://github.com/SensorsINI/jaer/issues/31

closed 2020-07-09 by julio-guillen