SensorsINI/jaer #31: Weird behavior when setting manually time in player

julio-guillen, 2020-06-10:
Hi,

As I cant use STEP BACKWARDS to navigate through my player, I'm trying to change time in player manually, but have weird behaviors:

https://pasteboard.co/JcpuwVa.png

Now jAER wont work, and I get this error in console:

https://pastebin.com/JyYjvQNQ

I've already tried to disable my filters, but now I cannot play any data.

Thanks in advance :)

tobidelbruck, 2020-06-10:
Thanks for reporting, I will try to reproduce the bug and push a fix. 
You should be able to get out of 3D mode by using "3" key or 
View/DisplayMode....

Does the error still occur in 2D rendering modes?

The stack trace is great, thanks for including it.

julio-guillen, 2020-06-10:
Hi Tobi,

I think the stack trace is not related. Thats why I edited it. It was just me painting something out ob bounds.

Sorry :)

julio-guillen, 2020-06-10:
Hi,

So the bug can be easily reproduced. Just edit manually time and press 3 :)

Thanks in advance

tobidelbruck, 2020-06-12:
Thanks Julien, thanks to your question I worked on this annoying lack  
and pushed some changes to enable backward forward stepping. The display 
will also show the actions. You can test it by pulling latest commits 
and rebuilding.

tobidelbruck, 2020-06-19:
backwards stepping fixed, now works. Also added yellow status for actions on display

closed 2020-06-19 by tobidelbruck

julio-guillen, 2020-06-25:
Works perfect. Great idea using yellow display.

Thank you very much.