SensorsINI/jaer #34: crash on opening file

martinosorb, 2020-09-01:
Hi all, I've simply tried to open a file ("Open logged data file..."), and jAER crashes immediately every time. I tried with different versions of aedat files. This is the console error:

```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00000001038e3f4e, pid=10392, tid=775
#
# JRE version: Java(TM) SE Runtime Environment (10.0.1+10) (build 10.0.1+10)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (10.0.1+10, mixed mode, tiered, compressed oops, g1 gc, bsd-amd64)
# Problematic frame:
# C  [libosxapp.dylib+0x2f4e]  -[NSApplicationAWT sendEvent:]+0x179
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /Users/martino/Lavoro/jaer-1.8.1.8/hs_err_pid10392.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
./jAERViewer_linux.sh: line 18: 10392 Abort trap: 6           java -classpath "$DIR/dist/jAER.jar:$DIR/jars/*:$DIR/lib/*" -splash:images/SplashScreen.png -Dsun.java2d.uiScale=2.0 -Djava.util.logging.config.file="$DIR/conf/Logging.properties" -Dsun.java2d.noddraw=true -Dsun.java2d.opengl=false -D-Jsun.java2d.dpiaware=true net.sf.jaer.JAERViewer "$@"
```

This is on Mac OS, with Java 10.0.1

Thanks

tobidelbruck, 2020-09-01:
Yeah, sorry about that!  For now forget about Java 9, 10, 11, etc.  Use 
Java 1.8 (java 8).  But on Mac OS there is an unfortunate bug or 
complexity in the Java bindings for Open GL (JOGL) that causes an 
alternation between projection transforms frame to frame, so there is an 
annoying flashing from black to correct on every pair of frames. We 
could never sort this out.

We are waiting for the next JOGL release (which has been a LONG time 
coming) to get everything working with Java >8.

For best experience, use either windows or linux. We use jaer on both on 
daily basis.

martinosorb, 2020-09-02:
Thank you. For other mac users, I use this script to launch jAER with the correct Java version:

```bash
export JAVA_HOME=$(/usr/libexec/java_home -v 1.8);
~/Software/jaer-1.8.1.8/jAERViewer_linux.sh```

You may not have version 1.8 installed, `/usr/libexec/java_home -V` will tell you the installed versions.

closed 2020-09-02 by martinosorb