SensorsINI/jaer #37: PatchMatchFlow - StartLoggingGlobalMotionFlows not working

rivetti22, 2021-02-15:
Hi there,

I'm trying to save the optical flow data to a file and from my understanding, it is done using the StartLoggingGlobalMotionFlows function. However, when I try to do so, I input the file name in which I want to store the data, and I get a "File not found" exception.
[image]
 exception.

I named the file "log_test", but as seen in the screenshot it appends date and time information to the original name I gave it. Can you please help me? I would be very thankful.

wzygzlm, 2021-02-16:
Hi Alessio,

Thanks for reporting the bug. This exception is caused by the upgrade for the TobiLogger. Patchmatchflow uses this logger to store data. Before the logger upgrade, the user has to specify a concrete path. Now we support to set to use the home folder as the default path. 

Anyway, this might not be important for end-user. I pushed a new commit to fix it. Check it.

rivetti22, 2021-02-16:
> Hi Alessio,
>
> Thanks for reporting the bug. This exception is caused by the upgrade for the TobiLogger. Patchmatchflow uses this logger to store data. Before the logger upgrade, the user has to [...]

rivetti22, 2021-02-16:
Hi Min,

First of all, thank you for the fast reply. I just tested it and it is working perfectly! Now I can just give a name to the file and it will store the logged data in that file, in the chosen path. Thank you for your help.

closed 2021-02-16 by wzygzlm