SensorsINI/jaer #38: java.lang.ClassNotFoundException: net.sf.jaer.JAERViewer

ShallowWill, 2021-02-22:
Hi, I cloned this rep and ran the jAERViewer_win64.exe. I already installed JDK in my Win system.

But I got the following error: 
java.lang.ClassNotFoundException: net.sf.jaer.JAERViewer
at java.net.URLClassLoader.findClass(Unknown Source)
at java.lang.ClassLoader.loadClass(Unknown Source)
at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
at java.lang.ClassLoader.loadClass(Unknown Source)
at com.exe4j.runtime.LauncherEngine.launch(LauncherEngine.java:77)
at com.exe4j.runtime.WinLauncher.main(WinLauncher.java:94)

any ideas for this problem? thanks a lot.

tobidelbruck, 2021-02-23:
Yes, you need to build the project. The compiled classes are not part of 
clone.

I guess the simplest would be to install Netbeans 8.2 and build the 
project, or you can install ant and build with ant (ant is like make but 
newer in some ways). The instructions should be in the user guide and in 
the README.md in github. Or see the youtube video.

ShallowWill, 2021-02-26:
thanks a lot, it solved the problem.

ShallowWill, 2021-02-26:
thanks Prof. Tobi, I want to synchronize DAVIS346 cameras with another device.

What I am looking at is ‘Input signals can be 3.3V or 5V, depending on the VDD_IN supplied externally, output signals are 5V, as is VDD_OUT. If you chain cameras together for synchronization, the clock and VDD will be 5V, for example.’, which is copied from the user guide of DAVIS346.

I am still confused about that. Could you please help to confirm the following:

(1) does it mean the voltage of SIGNAL_IN and VDD_IN should be the same, either 3.3v or 5V;

(2) do we need input for CLOCK_IN if we only sync one DAVIS346 camera with an external device? If yes, what the input should be?

(3) If we want to sync multiple DAVIS346 cameras, (as mentioned in the user guide: chain cameras), in this case, will CLOCK_IN and VDD_IN be 5V constant voltage?

Thanks a lot.

tobidelbruck, 2021-02-26:
Hi Will, Did you look at the guide? 
https://docs.google.com/document/d/1U7dQqRliF5-vg2YOSl2V2UB2W2TPEUDdSR5gHPANwqw/edit?usp=sharing

That google doc version might be out of date. The authoritative source 
would be inivation. It would make sense to post this kind of hardware 
question to davis-users forum, not to jaer-forum.

On 2/26/2021 10:43 AM, ShallowWill wrote:
> thanks Prof. Tobi, I want to synchronize DAVIS346 cameras with another
> device.
>
> What I am looking at is ‘Input signals can be 3.3V or 5V, depending on
> the VDD_IN supplied externally, output [...]
Yes, I think they should be same
>
> (2) do we need input for CLOCK_IN if we only sync one DAVIS346 camera 
> with an external device? If yes, what the input should be?
>
Yes, the clock in would carry the external input event signal, see the 
guide on configuring edge polarity, length, etc for this pulse.

> (3) If we want to sync multiple DAVIS346 cameras, (as mentioned in the 
> user guide: chain cameras), in this case, will CLOCK_IN and VDD_IN be 
> 5V constant voltage?
>
No, you just need 2 wire connector from master to each slave. It has 
ground and the 10kHz clock. See the guide above; I hope it makes it clear.

closed 2021-03-03 by ShallowWill