SensorsINI/jaer #40: can't build jaer on netbeans ide, verification always needed

lin88619973, 2021-04-12:
after clicking clean and build， it Pop up a dialog box just like this
verification needed
                  artifactory realm
username：
passport：

tobidelbruck, 2021-04-12:
It must be something to do with pulling the libraries using ivy from 
maven-central repository. I don't know the underlying cause and have not 
seen this particular problem.

Does it continue if you just enter null user and password?  Can you 
google maven-central / ivy authentication bug? What is the detailed 
logging output shown in the console?

lin88619973, 2021-04-17:
Thank you for your reply, I am using the 1.5 version of the platform recently, and it can be built normally. The version I currently use can meet my needs, and I will try your suggestions when I am free. Thanks again.

closed 2021-04-25 by lin88619973

julio-guillen, 2021-07-21:
I have the same issue. It's kind of annoying, as even when trying to delete the project, it still launches the user/password dialog box.

I've tried both version 19.5 and master versions with same results.

[image]

tobidelbruck, 2021-07-21:
Did I post solution to this?  I think it requires you to have a valid 
github token stored. But I might not be correct about my memory of this. 
It is some arcane thing related to access to the Maven repos for all the 
libraries.

julio-guillen, 2021-07-21:
Hi, thank you for the fast response.

Do you mean this [1] post?

[1] https://groups.google.com/g/jaer-users/c/nFIGoXOrO5Y/m/mhO2rDDaAwAJ

julio-guillen, 2021-07-22:
Hi,

Version 1.7.5 builds correctly.
Version 1.8.1.8 builds correctly.

Version 1.9.0 and greater requires authentication.

Regards

julio-guillen, 2021-07-22:
Hi,

If you constantly cancel each time Artifactory Realm dialog box shows, you will eventually be able to clean & build jAER. Kind of annoying, but it works.

Thanks

tobidelbruck, 2021-07-22:
Is your ivysettings.xml the same as HEAD version? I recall changing this 
line marked below to deal with this headache

  <credentials host="SERVER" realm="Artifactory Realm" username="USER" 
passwd="PASSWORD" />

tobidelbruck, 2021-07-22:
No, that post was for the post-tag release generation mechanism....

julio-guillen, 2021-07-22:
Hi,

Im using right now cloned version so ivysettings.xml contains that line. I've checked some of 1.9.x releases and they dont. So I guess that wont fix the issue.

Regards

tobidelbruck, 2021-07-22:
Do you have a stored github token?

tobidelbruck, 2021-07-22:
I.e. did you use the Netbeans Team/Remote/Fetch.... menu to store a 
login via token to github?

https://docs.github.com/en/github/authenticating-to-github/keeping-your-account-and-data-secure/creating-a-personal-access-token

julio-guillen, 2021-07-22:
Hi 

I just used git clone from command line. Ive tried Netbeans Team, using both a PAT and my own user/password, but same luck.

Sorry for all the trouble...

cargonriv, 2022-01-09:
Hello, 

I had the same issue for about a week while trying to build jAER on NB as well. It seems that adding a GlassFish server on your local domain seems to fix the problem for me. You can find the server's dialog box on the Tools tab under 'Servers'. I used the default localhost and ports, but I did add a username and password of my own. 

Hope this helps.