SensorsINI/jaer #42: UDP Remote Control issue

LoutishDuke, 2021-07-09:
Hi,

I'm having an issue using jAER across Windows, Mac, and Linux - I've tried version 1.7 - 1.9. The issue is that I am trying to use the remote control feature to start and stop logging using the user guide python example which does work; the code does in fact create a .aedat file but it is empty apart from the header information. When I activate the remote UDP input server for jAER using the tooltip in the menu the jAER AEViewer will freeze and I believe this is symptomatic of the issue as no events are being processed by the viewer to be logged in the file - once I stop the remote input through the tooltip the video/event feed unfreezes. I would appreciate any help with this.

tobidelbruck, 2021-07-09:
I can take a look, what are the steps to reproduce? I'm not quite clear. 
You are receiving events via UDP by AEViewer and also want to log these 
events using AEViewer controlled by a UDP connection, is that correct?

LoutishDuke, 2021-07-09:
Hi Tobi,

I've actually solved it myself by checking over some things and it's my own error - for context I want to drive logging DVS data via UDP from a DVS device connected to the localhost by USB - I didn't release that a UDP service is spun up automatically when jAER starts with port 8997 so I was creating a conflicting message, enabling the UDP functionality via the "FIle> Remote > Unicast etc." in jAER at the port obviously leads to this behaviour. Thanks for offering to assist anyway @tobidelbruck it is appreciated!

closed 2021-07-09 by LoutishDuke