SensorsINI/jaer #43: Can't run dvs-slice-avi-writer.sh, DvsSliceAviWriter main class not found

Omar-AV, 2022-03-25:
Hi there, 
I'm trying to run aedat_to_avi.sh, which calls the dvs-slice-avi-writer script. However, I get the error message, 

"Error: Could not find or load the main class net.sf.jaer.util.avioutput.DvsSliceAviWriter".

Indeed, there is no .class files if I go to the directory where DvsSliceAviWriter.java is stored, which makes me believe something goes wrong when building jAER. The machine is running Ubuntu LTS 20.04 with Java 1.8, accessed remotely through ssh. I have followed the user-guide by building jAER with ant to no luck.  Any suggestions what I'm missing? Do I have to build jAER with Netbeans while accessing the computer physically, like the jAER tutorials on Youtube perhaps?

tobidelbruck, 2022-03-25:
Let me check it. Thanks for reporting!

tobidelbruck, 2022-03-26:
I don't see any script named _aedat_to_avi.sh_.... is it part of jAER?

Did you look in the folder _build_? That is where compiled classes are placed after the ant "jar" target runs.
Actually all classes that are built should be placed into the jar archive after "jar" build.
The actual folder will be "_build/net/sf/jaer/util/avioutput/_"
Most likely the location changed and the linux script _dvs-slice-avi-writer.sh_ needs updating.

I can't debug that here from windows, sorry.

Omar-AV, 2022-03-28:
Thanks for the quick response! 
Aedat_to_avi.sh is not part of jAER(I believe), it makes calling *dvs-slice-avi-writer.sh* on a large amount of *.aedat* files easier. It's also written by SensorsINI and can be found [here](https://github.com/SensorsINI/processAEDAT/tree/master/aedat_to_LMDB).

As you mentioned, the *build* folder does indeed hold all the compiled .class files, thanks for pointing that out. Sadly, adding the *build/net/sf/jaer/util/avioutput/* path after the *-classpath* flag in *dvs-slice-avi-writer.sh* still returns the same error.
I do have some other questions in order to try debugging this some more, if you have the time and don't mind.

- You mention in the jAER youtube tutorial, video 1, between time [1:04-1:15](https://youtu.be/8PSu07n8qCw?t=63) that running *jAEViever.exe* in windows sets up the java classpath. Do you know if one must run the equivalent *jAERViever_linux.sh* once to setup the classpath on linux? Is this relevant or should the *-classpath* flag in *dvs-slice-avi-writer.sh* be enough?
- Regarding .jar-files, I've looked at the README under jaer/jars and it mentions that there should be a linux32 and linux64 directory on linux platforms that holds native libraries for the platform. I don't have such directories, is this maybe fixed by running *jAERViever_linux.sh* and could my issue stem from the lack of such folders? 
- On the note of .jar-files, are there important ones I should be aware of that might have new locations I should try to modify in the *-classpath* flag of *dvs-slice-avi-writer.sh*? 
- Finally, I mentioned that I'm running Java 1.8, but it's OpenJDK/JRE-8. Mainly because Oracle JDK 8 is developer only and can't be obtained without provided access. Is jAER compatible with OpenJDK, or should one stick with a newer version of Oracle JDK instead?

tobidelbruck, 2022-10-11:
I'm having difficulty compiling jAER in WSL2 which is the only linux I 
have easy access to. I will try it in windows to see what's going on.

tobidelbruck, 2024-01-12:
Issue apparently solved, but should check dvs-slice-avi-writer after jaer-2.0.2 version is current release

closed 2024-01-12 by tobidelbruck