SensorsINI/jaer #44: No response when running

loliq, 2022-08-23:
Hi,I have compiled jaer1.8.1, during building process, every setup is following the YouTube setup video. But when I start running, there is no response. 
[image]

And I have tried to use the compiled exe, which is download from [here](https://github.com/SensorsINI/jaer/releases/download/1.8.1.8/jaer-1.8.1.8.zip).

The window has no response like : 
[image]

Looking forward to your reply.

tobidelbruck, 2022-08-23:
You mean it gets stuck at the splash screen?
Could you supply a  bit more information about your computer, OS, which JDK you are using?
I've been developing jAER robot for past few weeks with no problem on windows 10 laptop.

You can also run under debugger and try to see which thread is stuck where. The main animation loop thread is AEViewer.

loliq, 2022-08-24:
> You mean it gets stuck at the splash screen? Could you supply a bit more information about your computer, OS, which JDK you are using? I've been developing jAER robot for past few weeks with no [...]

Thanks for replying. And yes, it gets stuck at the splash screen.  My OS is Windows 10. JDK is downloaded from the User Guide, which is [jdk-8u333-windows-x64.exe](https://drive.google.com/file/d/1RDEmByCxxcGlgP0pFXFfCEcba-CsKFU9/view?usp=sharing). I tried to debug ,it also gets stuck and doesn't output any useful information like:
[image]

tobidelbruck, 2022-08-24:
Hmm, can't see anything obvious. One thing that is a bit strange is that you are getting a splash screen even though the log says there is no splash screen. 

Are you running with debugger in netbeans?
If so, maybe you could step jaerviewer to see where it gets stuck. It will be somewhere after the last logging output.

-- 
Tobi Delbruck, www.ini.uzh.ch/~ tobi, Swiss cell +41766291500, USA cell +16265102646

julio-guillen, 2023-04-16:
Hi,

I had the same issue. The problem was that my java version was java 8 update 361 and it didnt work. I downgraded to update 333 and it worked correctly again.

Bit late but hope it helps.

tobidelbruck, 2024-01-12:
Closed without clear resolution. Please try jaer-2.0.2

closed 2024-01-12 by tobidelbruck