SensorsINI/jaer #45: When I opened jAERViewer_win64.exe, it occured these errors. why? I have installed jdk 1.8.

LI474525323, 2022-11-09:
[image]
java.lang.ClassNotFoundException: net.sf.jaer.JAERViewer
	at java.net.URLClassLoader.findClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at com.exe4j.runtime.LauncherEngine.launch(LauncherEngine.java:77)
	at com.exe4j.runtime.WinLauncher.main(WinLauncher.java:94)

julio-guillen, 2023-04-16:
Hi,

This is a weird one. JVM says that he is not finding  the class "net.sf.jaer.JAERViewer" when trying to launch jAER. I think you should provide more info in order to understand better the situation.

Regards

tobidelbruck, 2024-01-12:
Please try the install4j insttaller. It comes bundled with Java and all the libraries and can be converted to .git working copy from the Help menu.

closed 2024-01-12 by tobidelbruck