SensorsINI/jaer #48: JAER problem with the EVK4 camera outputs

azemrani, 2024-01-18:
I am trying to use JAER for an event-based camera (EVK4) to perform noise reduction. However, when I open the sample dat files of this camera (https://docs.prophesee.ai/stable/datasets.html) using jAER, it does not show anything and didn’t work. 

[image]

wzygzlm, 2024-01-18:
Hi,

jAER hasn't supported the Prophesee cameras and its data format yet.

azemrani, 2024-01-18:
Thanks for the guidance. Is there anyway to convert Prophesee dat files to the formats supported by jAER?

tobidelbruck, 2024-11-21:
sorry for late answer. The simplest route is to convert to text CSV format, which jaer can read natively.  I don't know anything about prophesee format or of tool to do this but it must be possible.

Then in jAER add the AEChip class [DVS1280x720SD](https://github.com/SensorsINI/jaer/blob/master/src/ch/unizh/ini/jaer/chip/retina/DVS1280x720SD.java)

[image]

See [Data file formats in jAER user guide](https://docs.google.com/document/d/1fb7VA8tdoxuYqZfrPfT46_wiT1isQZwTHgX8O22dJ0Q/edit?tab=t.0#bookmark=id.9e7qp8ppuu4c) and [YouTube video Playing CSV files in jAER](https://www.youtube.com/watch?v=I5jdMzXWrbU)

closed 2024-11-21 by tobidelbruck